LibrePCB/librepcb-rfcs #1: Add interactive push and shove router from CERN/kicad/horizon

kliment, 2018-02-09:
Having reviewed the code and horizon's integration, this should be quite easy to add and gives a lot of useful functionality. If nobody objects, I'd like to start work on it as soon as the board editor is in a usable state.

ubruhin, 2018-02-10:
Yes, that would be great. Do you know if the router also supports a very simple routing mode which just creates the traces where the user clicks, without any obstacle detection and so on? I mean, similar to the "router" which is currently implemented in LibrePCB. If so, we could completely remove our own router and replace it by the kicad router.

Btw, If you have some links to code and/or documentation of the kicad router, could you post them here so we have quick access to the most important information?

dbrgn, 2018-02-15:
> Do you know if the router also supports a very simple routing mode which just creates the traces where the user clicks, without any obstacle detection and so on?

Yes, I think so. The interactive router has three modes:

[image]

If you use "highlight collisions", then no obstacle avoidance is being done.

ubruhin, 2018-08-08:
I started a first attempt with KiCads PNS router, just to get a first impression how much effort is required to integrate it. Still hard to say, but at least I already had a first "hurray"-effect :smiley: 

[image]

The code is hacky and buggy like hell, and it doesn't work at all (nothing gets really connected), but I still like it already :grin:

dbrgn, 2018-08-09:
Woahh, that is fantastic! I didn't expect an initial integration attempt that fast :tada: 

> Actually I wonder whether I should properly implement the current routing tool to make it usable or it's better to completely replace the current routing tool with the PNS router. If the PNS router also supports a simple routing mode, and the effort to integrate it is similar to the effort needed for improving the current router, then it would make sense to directly integrate the PNS router...

I really like the Kicad router, so I'd be :+1: for that.

rnestler, 2018-08-13:
> I started a first attempt with KiCads PNS router, just to get a first impression how much effort is required to integrate it. Still hard to say, but at least I already had a first "hurray"-effect smiley

Do you have this initial version on some branch?

ubruhin, 2018-08-13:
> Do you have this initial version on some branch?

Yes: https://github.com/LibrePCB/LibrePCB/tree/add-pns-router

dbrgn, 2018-08-13:
Since at least 3 projects use this router now (KiCAD, Horizon, LibrePCB) it might be interesting to extract it into a separate repository.

Maybe we could suggest this to the original authors once the PoC is done.

ubruhin, 2018-08-13:
> Since at least 3 projects use this router now (KiCAD, Horizon, LibrePCB) it might be interesting to extract it into a separate repository.

Yes, that would really be great! :+1:

> Maybe we could suggest this to the original authors once the PoC is done.

Or even *now*, I don't see a reason why to wait ;)

Someone interested to do it? @kliment?

kliment, 2018-08-13:
The original authors are CERN, as part of the KiCad project. Since they're in the middle of their most important release ever I doubt this is a good time to ask them to do major code base restructuring.

dbrgn, 2018-08-13:
That's not a major code base restructuring. The router is quite standalone, so it shouldn't be too hard to extract it into a separate repository.

I'll take a look if I find a developer forum or mailing list to ask :)

kliment, 2018-08-13:
The people in charge of the PnR are Tomasz Wlostowski <tomasz.wlostowski@cern.ch> and Maciej Suminski <maciej.suminski@cern.ch>

dbrgn, 2018-08-13:
@kliment thanks! :)

dbrgn, 2018-08-15:
Reply: At the moment they're not interested in extracting the code into a standalone library. That's fine, we can simply copy the router (it's GPLv3+ licensed).

rnestler, 2018-10-29:
> So it would be great if someone would start creating such a standalone library :)

I can give it a try. I'll do it in my personal GitHub account first, we can then still move it to some organization if needed.

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/add-interactive-push-and-shove-router-from-cern-kicad-horizon/13

closed 2018-12-14 by ubruhin