LibrePCB/librepcb-rfcs #6: Snap grid issue

ouabache, 2018-03-17:
Kicad has a recurring problem where a user tries to place a component created under one snap grid spacing onto a schematic sheet with a different snap grid. LibrePCB has the same issue. We need to make it impossible for this to occur.

One solution would be to save the snap grid spacing in every symbol when it is created and compare it with the schematics grid when it is placed. If they differ then divide the symbols position points with its snap grid spacing and then multiply by the schematics spacing. The graphics might look a little funky but all the pins will connect.

Another solution is to get rid of the snap grid spacing control all together. 

John Eaton

dbrgn, 2018-03-17:
Hm, yeah, I know that problem. It's really annoying, especially since some components have metric pin spacing and others have imperial ping spacing.

I could imagine a greatest-common-divisor style approach could work. But there should probably be a way to turn it off in case someone actually wants to follow the snap grid exactly.

ubruhin, 2018-03-18:
I think we should not compare the grid interval of the symbol and the schematics, because you could still place pins/pads off the grid (you can specify arbitrary coordinates). In that case it wouldn't help to compare the grid intervals.

The problem is actually only in the editor state machines which handle the mouse clicks. Their logic should be changed to completely ignore the grid interval when searching for pins/pads under the cursor, to make snap working even if they are off the grid. And probably some more adjustments are needed to ensure only orthogonal lines are created when drawing netlines/traces...

ouabache, 2018-03-18:
The problem with ignoring the interval when searching for pads is that sometimes you place the wires down first and then drop the component on top of them. If your pins are off grid then that would be a nightmare.

John Eaton

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/snap-grid-issue/33

closed 2018-12-14 by ubruhin