LibrePCB/librepcb-rfcs #7: Documentation crosslinking structure

hephaisto, 2018-03-17:
When I started to read about LibrePCB, it took me quite a while to find all necessare resources. For example, I haven't found any link to the doxygen files on the README.md. They link to something referred to as "documentation", but what is rather a "user manual" - at least when I read "documentation", I would expect something a lot more technical and detailled. I actually found out how to build the docs myself before I found out where to find the generated doxygen files online. It actually took me about 10 minutes to find it and it only worked to google "librepcb doxygen" (for the record, its here: http://librepcb.org/LibrePCB-Doxygen/master/ ).
I would suggest to modify occurrences of "documentation" to something like "manual" and add a link to the doxy files on a few central pages (README.md of main repo, librepcb.org, ...) under the name "documentation".

I think it is obvious that a link to the doxy files is important, but would you concur with the proposed labelling above?

ubruhin, 2018-03-18:
Makes sense to me. So I would summarize it in two points:
- Rename "documentation" to "user manual" (in README.md)
- Add link to doxygen documentation (in README.md)

And maybe a third point:
- Remove "doc" directory (move images to "img/") because there is no documentation in the main repository (doxygen is generated, and manual is a separate repository)

hephaisto, 2018-03-18:
I'm not sure whether the images in the doc directory really belong in the main repository. Maybe it would be better to put it in LibrePCB-doc.

I also think that a more verbose link label would be good. Something like:
* [User Manual](https://docs.librepcb.org/)
* [Automatically generated documentation (doxygen)](http://librepcb.org/LibrePCB-Doxygen/master/)

ubruhin, 2018-03-20:
> I'm not sure whether the images in the doc directory really belong in the main repository.

These are only images used in the README.md of the main repository.

> I also think that a more verbose link label would be good.

:+1:

ubruhin, 2018-03-21:
Fixed by @hephaisto in https://github.com/LibrePCB/LibrePCB/pull/244.

> Remove "doc" directory (move images to "img/") because there is no documentation in the main repository

This could still be done some day later...

closed 2018-03-21 by ubruhin