LibrePCB/librepcb-rfcs #8: inmutable flag

ouabache, 2018-03-18:
librePCB has a deprecated flag in each component which is important to know but it also needs an
immutable flag as well. Mutable components are still under construction and can change without
notice. Immutable means that it is frozen and will never change, any errors will be corrected in the next revision and will be give a new version number.

This also means that each component needs a version field as well.

BTW: The immutable flag belongs in the symbol file because the owner of that file determines whether the design is frozen. The deprecated flag is determined by the end user and will change on a case by case basis. 

There are parts that I cannot use for production but I am allowed to use for a temporary test board. The
deprecated list changes depending on what you are working on.

John Eaton

ubruhin, 2018-03-18:
I don't think that an immutable flag is required, because when using a library entity (component, symbol, ...) in a project, the entity is copied into the projects directory (*not* just referenced). So any further changes to the entity in the library will *not* have any effect to existing projects, and thus it's always safe to make updates on library entities.

> This also means that each component needs a version field as well.

That's already the case.

> The deprecated flag is determined by the end user and will change on a case by case basis.

The deprecated flag is intended for library entities which are replaced by another entity, so in my opinion it's correct that the flag is stored in the entity file itself. Maybe you mean something more like a stock management (active parts, NRND, no longer available, ...)? This would be a completely different topic...

ouabache, 2018-03-18:
I was thinking of the case of a design team where one designer will make a release candidate available for the rest of the team to begin working with it. You do not want this in your projects directory because you might miss the final release and become out of sync with the rest of the team. The immutable flag warns you that it will be changing.

I was thinking about deprecated as a NRND flag.

John Eaton

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/inmutable-flag/42