LibrePCB/librepcb-rfcs #9: Download libraries manually doesn't work

ouabache, 2018-03-19:
There is a 60 char length limit on the name. Most URL's are longer

dbrgn, 2018-03-19:
The URL is not limited, but the "local directory" path is. This is what you meant, right?

[image]

I'll create a pull request to fix this.

dbrgn, 2018-03-19:
By the way, this is something that could be reported to the LibrePCB repository. This repository is meant for discussion of larger features, not for bugreports.

ubruhin, 2018-03-19:
I don't understand why this is an issue:
- The URL is not limited to 60 char
- The "local directory" is *not* a full path, but only a directory *name*. I think it doesn't make sense to create a directory with a name longer than 60 characters...

ouabache, 2018-03-19:
OK My bad. I ran into this yesterday but now I can no longer reproduce it. I'll let you know if I can figure what was going on.

John Eaton

ubruhin, 2018-03-19:
OK I'll close this issue. Just let us know if it's still an issue.

closed 2018-03-19 by ubruhin