LibrePCB/librepcb-rfcs #11: Human Readable Library Paths

ouabache, 2018-03-21:
The use of UUID's as directory names is not human friendly at all. 

librePCD.org__Resistors__resistor  is as unique as  75372c18-3ba4-42e8-b3b2-2eb5039d4
and a lot easier to follow.

Anytime a UUID is presented to the user it should be mapped back to its author,library and component names.

John Eaton

hephaisto, 2018-03-21:
I agree that it is not user-friendly when working on the files manually.

However, as far as I understand, the general idea is to reference *anything* by its UUID, so it is consistent to do so also on the directory level. Also, there would be need for a big lookuptable that maps some arbitrary tuple of attributes to an UUID.
If this is really a big deal for usability, it would be easy to write a script that creates symlinks from a name to an UUID.

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/human-readable-library-paths/30

closed 2018-12-14 by ubruhin