LibrePCB/librepcb-rfcs #12: Variant management

ubruhin, 2018-03-24:
_From @timhoeppner on March 23, 2018 23:1_

Came across this project from a colleague at work and watched the video from FOSDEM. I've got to say, this project shows incredible potential. I think you nailed it on the head about why you need to start from scratch regarding library management and how it is so core to the software. It's also refreshing to see the use of modern tools for development making the development barrier much lower than other open source CAD tools...

That aside, one thing that I find missing in many of the CAD tools is variant management. This more of a professional level feature but is something that I would really like to see in a set of free tools. My only other experience with a similar feature is in OrCAD CIS which is where I'm basing a lot of my ideas from. In case its not clear what I mean. I'm talking about configuration control for which parts should and should not be populated on a PCB. This would tie into BOM output as well as schematic generation. From my experience, we often develop PCBs that include certain features that can be selectively populated depending on the product model. This could also include having a variant for development and test which includes a bunch of test points that wouldn't otherwise be included on a production assembly.

Here are a number of key points this feature would include:
1. Separate window for managing variant configuration
   * Two-column layout
   * Left-side: Hierarchical tree view of variants, top-level items would be ones that you can actually generate a BOM or schematic from.
   * Right-side: List of parts associated with the selected variant. You would be able to organise which parts are populated or not in this view.
   * Variants can be dragged and dropped on the left side view to become children of a different variant
   * Variants can have a selection of parts and doesn't need to include all parts in the design. This is useful to define parts that belong to a certain feature within the design.
2. All designs would start with a default variant called "All" and all parts would be marked as populate.
3. Context menu in the schematic editor for quickly marking components as populated or not populated for a specific variant. Multiple parts could be selected at the same time for this.
4. The schematic editor would include a drop down to view the schematic with a certain variant applied. All components that are marked as not populated would appear greyed out for easy identification.
5. PDF export of the schematic would automatically create PDFs for all top-level variants.
6. BOM export would automatically create BOMs for all top-level variants.

I think that's the gist of it. This is something that I would like to contribute to the project so I'm really just looking for some feedback.

_Copied from original issue: LibrePCB/LibrePCB#245_

ubruhin, 2018-03-24:
_From @dbrgn on March 24, 2018 11:27_

There is a feature something a bit in this direction: You can "clone" a PCB and create multiple PCB variants of the same schematic.

I think you want something a bit different though: You want just one PCB design, but with some parts not populated, correct?

Sounds like an interesting idea :) I'd have some use for that as well.

ubruhin, 2018-03-24:
> I've got to say, this project shows incredible potential.

Nice to hear, thanks! :smiley: 

> That aside, one thing that I find missing in many of the CAD tools is variant management.

Yes, I fully agree. Actually I already planned to implement variant management some time. And probably I would have implemented it very similar how you described it :smiley: 

So generally I really like your proposal. I have just one question: What are the nested (non-toplevel) variants for (i.e.what's the use-case of them)?

In addition, as @dbrgn wrote, LibrePCB already has the ability to create different boards for the same schematic. In the different boards one can even choose different devices for the same component (e.g. 0603 package on one board, and 0805 package on another board). This feature probably needs to be combined with variant management as both do similar things.

ubruhin, 2018-03-24:
_From @timhoeppner on March 24, 2018 14:15_

> There is a feature something a bit in this direction: You can "clone" a PCB and create multiple PCB variants of the same schematic.

I noticed that and saw the demonstration in the video. Definitely an interesting feature on its own. The feature I'm describing is about BOM management which is really independent of the PCB. However, in this case, since you can select different footprints in the PCB and have different variants of the PCB, this would need to tie into that.

> This feature probably needs to be combined with variant management as both do similar things.

Agreed. This certainly creates some challenges and so I'm wondering the best way to integrate these. Wondering if it would make sense to limit PCB variants to the toplevel variants on the schematic. Ultimately you need to generate a BOM for your design and so I think these goes hand in hand.

> What are the nested (non-toplevel) variants for (i.e.what's the use-case of them)?

This is a feature available in OrCAD CIS. What I used it for is grouping off sections/features of the schematic. The nested levels could be copied to other top level variants for quick and easy management.

Just thinking about this now, I'm wondering if it would make more sense to call the top-level elements "variants" and sub-level elements "features". Not sure I've ever seen more levels then that anyway or can even think of a use case. In this sense, you can define "features" and just assign them to one or more "variants".

I think it might be useful for me to draft up a diagram that shows this a little more clearly.

ubruhin, 2018-03-24:
Yeah, the [Issue Mover for GitHub](https://github-issue-mover.appspot.com/) seems to work :grin: I moved the issue to this repository because variant management is a pretty "big" feature which probably needs some discussions to clarify all details until it can be implemented.

> I think it might be useful for me to draft up a diagram that shows this a little more clearly.

That would be great!

ouabache, 2018-03-24:
Be careful that you are not talking about 2 different things.

Platform design is where you create one schematic for one PCB  but have multiple  BOM's and solder masks. Each BOM/Mask pair creates a variant of the product. The solder mask is needed because you can configure a board using solder jumpers.

You would want to be able to take each BOM set and apply it to the master schematic so you could have a separate plot of each variant.

Variant design is where you have one master schematic that can produce a different PCB for each variant. This is done by changing a parameter and creating a different board.

We need to support both

John Eaton

ouabache, 2018-03-25:
We should for the record list all of the different types of designs that librePCB will support. My suggestions:

Simple           The Design is on a single sheet and produces a single PCB

Multi-sheet    The Design is spread over multiple sheets that together create a PCB.

Hierarchal      A top level sheet can instantiate symbols of other sheets to create a PCB. This is recursive

Beyond that each of these methods can be expanded:

Variant        The master design uses parameters and any number of derived variant PCBs can be created by changing the parameter set

Platform        The master design creates a single PCB and its function changes by changing the BOM or solder mask

Note that in both cases you will have a documentation package for the master and a second package customized for each variant

And finally we have:

Alternate    A single board design can create multiple PCBs. For example you may lay out both 2 layer and 4 layer versions of the same board or change packages for a part.

John Eaton

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/variant-management/38

closed 2018-12-14 by ubruhin