LibrePCB/librepcb-rfcs #16: Plan with Libraries

lukabosnjakovic, 2018-06-08:
I know that LibrePCB is in high development stage but is there any plan for Libraries? Will they be imported from some other project or is plan that we will make it on our own? If we are doing them on our own maybe is good idea to open repo only for libs. And it would be wise to make some policy for accepting components others made

ouabache, 2018-06-08:
If LibrePCB succeeds then we can expect that libraries will develop the same way they did with kicad. We start by supplying some basic components and then continue to add to the official releases. Then other groups like Digikey and sparkfun will develop their own libs and make them available from the internet.

Having a tool that can convert a library from kicad ,eagle gEDA etc into LibrePCB would save a lot of time when importing other designs into LibrePCB.

John Eaton

ubruhin, 2018-06-08:
There is a simple importer for Eagle libraries, but it's not (yet) polished to make it available for end users of LibrePCB. So currently the way to go is to create our own libraries, and accept contributions from others.

> If we are doing them on our own maybe is good idea to open repo only for libs.

Here they are: https://github.com/LibrePCB-Libraries

> And it would be wise to make some policy for accepting components others made

Yeah in #13 we discuss what library design recommendations we should make. It's planned to document the recommendations in https://github.com/LibrePCB/librepcb-doc, but nobody started with it... So feel free to start documenting it :)

ouabache, 2018-10-31:
I am working on a perl script that can convert a kicad library into LibrePCB. I have done the 10% effort that gives 90% results and can create symbols for most of the kicad and digikey libraries.

Should these be placed in the official LibrePCB repository? I intend to go over the license files to make sure that my copies comply but some of them require statements in the files themselves. Does LibrePCB lib files allow comments?

Has anyone done any work on a library coding standard or documentation for the file formats?

John Eaton

ubruhin, 2018-10-31:
Aren't all official KiCad libraries published under the [KiCad Libraries License](http://kicad-pcb.org/libraries/license/)? If yes, I'm not sure if we can (or should) convert and publish them as official LibrePCB libraries. My goal is that LibrePCB users can do *absolutely everything* with our libraries, without even worrying about their license. IMHO this requires to put our libraries under the CC0 license, which would mean we can't use KiCad libraries :(

See also discussion about KiCads 3D models here: https://github.com/LibrePCB/librepcb-rfcs/issues/32#issuecomment-433195649

> Does LibrePCB lib files allow comments?

No, that's (currently) not possible because they would be removed when saving a library element which contains comments.

> Has anyone done any work on a library coding standard or documentation for the file formats?

Not really... See https://github.com/LibrePCB/librepcb-doc/pull/18

ouabache, 2018-11-01:
So converting any other groups libraries without their explicit consent is probably not a good idea. 

I have run into a few problems converting kicad symbols into LibrePB. A lot of kicad and digikey libraries use a pin name of "~" for gates where the function is obvious. They can do this because they also have the pin number to keep track of the pin. We don't so any conversion script would need to generate a pin name and it would be hard to come up with something meaningful.
I am also having a hard time seeing how they handle multi-gate parts. I may have to dig into their code to see how they figure it out.

But it has been a useful test to see how the library manager handles a 8,000 part library. There is some slowing but nothing excessive. 

John Eaton

ubruhin, 2018-11-01:
> So converting any other groups libraries without their explicit consent is probably not a good idea.

Yes, I think so.

> A lot of kicad and digikey libraries use a pin name of "~" for gates where the function is obvious. They can do this because they also have the pin number to keep track of the pin. We don't so any [...]

Maybe it would be reasonable to just name them with incrementing numbers (i.e. "1", "2", ...)? They can still renamed after a manual review if needed, as it doesn't break something.

> But it has been a useful test to see how the library manager handles a 8,000 part library. There is some slowing but nothing excessive.

Good to hear :smiley:

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/plan-with-libraries/25

closed 2018-12-14 by ubruhin