LibrePCB/librepcb-rfcs #18: Setting development envirement

lukabosnjakovic, 2018-06-17:
I'm trying to start doing fixes but thing is that I want to pull my hair out :D 

I switched from Windows to Linux two months ago so if I ask something trivial please have understanding it is not trivial to me.

Ok, I've done fork and clone git repo to my PC. 
First I tried to start coding with VS code (_yes I know I'm stupid Windows user that worships Microsoft and Visual Studio, but that is because I like it a lot and one  big lack to me on Linux is that there is no VS for Linux, VS Code is only text editor behalf VS Comunity on Windows_) and I find out that it won't go as I planed. 

So I installed Qt Creator, open project and when I'm trying to start debugging I get "_**Cannot find file: /home/luka/Radno/LibrePCB/libs/delaunay-triangulation/delaunay-triangulation.pro.**_" error. 

Can someone help me? I don't know did I forget something (here is 00:23h and I'm little bit tired so maybe is that problem) or I didn't install all dependencies (if so where can I find dependencies list) or what?

Thanks :)

ubruhin, 2018-06-18:
Hi @lukabosnjakovic, you probably just forgot the `--recursive` flag when cloning the repository, so the submodules are missing :wink: 

See https://github.com/LibrePCB/LibrePCB/blob/master/README.md#cloning

Hope this helps :)

closed 2018-06-18 by ubruhin

dbrgn, 2018-06-18:
You can also do `git submodule init` followed by `git submodule update` inside the repository to get the submodules.