LibrePCB/librepcb-rfcs #19: Loading connectivity objects from a library

ouabache, 2018-06-19:
One thing that gEDA's gschem does that we should copy is the ability to define bus rippers in a symbol file and load them into a schematic from a library. The alternative is to synthesize the graphic in the program but that is undesirable because you are hard coding that into your code. This makes it difficult to change without recompiling the code.

This should be done for all connectivity objects: Bus rippers,symbol pins,io_ports,junctions etc. This would give the user more control over the look of their schematics and would simplify the tools.

John Eaton

ubruhin, 2018-06-20:
Sorry, I really don't understand what you mean ;) Could you give a concrete example, maybe with screenshots?

ouabache, 2018-06-21:
gschem has a section in its system-gschemrc file where you can select the busripper from a choice of
busripper-1.sym or busripper-2.sym that are simply two symbol files. This makes it easy for the user
to create and add their own busrippers if they do not like the default ones.

This could be extended  to also include symbol pins, io_ports, power and ground connectors,junctions etc.  I think it would give the users the ability to control the look of their drawings without requiring a lot of support.

John Eaton 

system-gschemrc
------------------------------------------------------------------------------------------------------------------
; Bus ripper controls
; The following keywords control the auto bus ripper addition code
;
; bus-ripper-size  : Sets the size of the auto bus rippers.  
; bus-ripper-type  : Sets the bus ripper type either a "component" or 
;                    plain "net"
; bus-ripper-syname  : If above is set to component, specify the symbol name.
;                      The symbol must exist in a component library
; bus-ripper-rotation  : Either "symmetric" or "non-symmetric".  This deals 
;                        with how the bus ripper symbol is rotated when it 
;                        is auto added to a schematic.
;   

; The default bus ripper
(bus-ripper-size 200)
(bus-ripper-type "component")
(bus-ripper-symname "busripper-1.sym")
(bus-ripper-rotation "non-symmetric")

; A symmetric alternative 
;(bus-ripper-size 200)
;(bus-ripper-type "component")
;(bus-ripper-symname "busripper-2.sym")
;(bus-ripper-rotation "symmetric")

; A simple net 
;(bus-ripper-size 200)
;(bus-ripper-type "net")

hephaisto, 2018-06-21:
> This could be extended to also include symbol pins, io_ports, power and ground connectors,junctions etc. 

Isn't this already covered by having different symbol variants?

About the bus-ripper: Is there even support (planned) for buses? I haven't seen them in LibrePCB, and to be honest, I've never seen bus symbols in actual schematics I used. People tend to just label stub net lines and skip the bus-drawing step entirely.

ouabache, 2018-06-21:
Isn't this already covered by having different symbol variants?
-----------------------------

Yes, and as long as the user can create their own symbols and use them then we are ok. But if you have a menu pick to add something then you want to give the user the ability to configure that pick to use one of their symbols.

Bus symbols are useful for larger designs. Small flat designs can manage fine without them but larger designs need hierarchy and that calls for bus-symbols.

All schematic capture tools that I have ever used support homogeneous and some even support heterogeneous bus-symbols that let you do huge designs. Kicad is going to add support for heterogeneous buses either in V5 or V6. If librePCB wants to be taken seriously as a tool then you should support both types of buses.

John Eaton

ouabache, 2018-06-21:
> Is the second sentence purely hypothetical or are you referring to an actual or planned feature?

That's what gEDA does. It has a menu pick for inserting bus-rippers but lets the user decide which symbol to use

>You are kind of missing the point of what I want to say. The question is not whether support busses, but rather whether to have a graphical symbol representing it.
I still don't really get what you actually want to achieve. I'm mostly used to eagle, which just lets you connect nets to buses.

I see your point now. You don't need a symbol for a ripper between a net and a homogeneous bus if the net name already matches the homogeneous bus member name but otherwise you need one. This is especially true with heterogeneous buses where the ripper is used to map signal names.

John Eaton

ouabache, 2018-06-22:
Another thing is that if you have a bus ADDR[31:2] then you can connect a net ADDR[2] directly to the bus without a ripper and it will connect by name. But bit 2 is really bit 0 of a physical 30 bit bus. A ripper lets you show the actual index value as well as the logical name.

hephaisto, 2018-06-23:
Isn't it normal, that nets with the same name are connected? So if you have a net name `ADDR[2]` whereever you defined the "bus" (e.g. at the CPU), having a second net segment with the name `ADDR[2]` (e.g. at a RAM block) should already connect these nets, right? From this perspective, creating a "bus" is just a matter of optics and documentation and could be achieved with a simple drawing tool.
I'm not really familiar with the workflow when drawing such connections with a real drawing tool. I usually make all connections between different modules (logical groups) by giving net stubs the same name. This way, by allowing special characters like `[` in net names, this would already be covered by the (hopefully) currently working net handling tools.

ouabache, 2018-06-23:
> Isn't it normal, that nets with the same name are connected? So if you have a net name ADDR[2] whereever you defined the "bus" (e.g. at the CPU), having a second net segment with the name ADDR[2] [...]
I'm not really familiar with the workflow when drawing such connections with a real drawing tool. I usually make all connections between different modules (logical groups) by giving net stubs the same name. This way, by allowing special characters like [ in net names, this would already be covered by the (hopefully) currently working net handling tools.

You could do a schematic with no wires or busses at all. Simply add a small wire stub to each pin and tag it with a net name and you have your connectivity. That is a graphical net  list and is pretty useless in conveying the big picture of what is going on in the circuit. The reason we do schematics is to show the viewer the total picture of what is going on in the circuit and for that you need wires and busses.

Its not enough to simply put a graphic line on your sheet but still connect via names. Adding the line must create the connection and deleting it must break it. Otherwise it is hell trying to keep the graphics in sync with the connections

John Eaton

ubruhin, 2018-06-23:
Hmm ok the idea is interesting, but to be honest, I'm not yet sure whether connectivity objects should really be user-defined or built-in (as currently)... Anyway, I'll keep this idea in mind.

> Is there even support (planned) for buses?

Not yet implemented, but planned. I think a professional EDA tool needs support for buses (*how* they are implemented is a different topic).

> All schematic capture tools that I have ever used support homogeneous and some even support heterogeneous bus-symbols that let you do huge designs. Kicad is going to add support for heterogeneous [...]

Good point. I created #20 to keep track of this topic.

ubruhin, 2018-06-23:
> One of the classic rules in programming is "Never hard code constants in your program"

Sure. But connectivity objects are much more complex that a magic number ;) There are many things to consider (layers, where to store them, how to choose them, how to draw them...) and thus it would require considerable effort to implement it properly.

On the other side, I guess the only built-in connectivity objects are currently symbol pins and the connection point of connected lines. Both are extremely trivial graphical objects (line + circle). I wonder why one wants to change these simple objects (which other look would be useful?).

I totally agree to not make supply symbols like GND or VCC built-in because there are many different looks possible, and people have different opinions how they should look. Some tools seem to have these symbols built-in, but LibrePCB not. They are normal library symbols like resistors and so on, it was pretty easy to implement it that way (in opposite to connectivity symbols, which are a quite different thing).

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/loading-connectivity-objects-from-a-library/29

closed 2018-12-14 by ubruhin