LibrePCB/librepcb-rfcs #21: Symbol placement: Rotate with right-click

dbrgn, 2018-07-17:
When adding new symbols to a schematic, the right-click right now does not seem to do anything.

For some reason, I think it would be intuitive to rotate the part using the right mouse button. Opinions?

ubruhin, 2018-07-17:
Actually that's already implemented, and it works on my system. Could you try if it works with the nightly AppImage?

dbrgn, 2018-07-18:
> Actually that's already implemented

Ah, that's why it seemed intuitive to me :smile: 

I found the source of confusion though. It does work in the schematic, but not in the symbol editor of the library editor (e.g. to rotate a pin).

ubruhin, 2018-07-20:
Moved to the main repo: https://github.com/LibrePCB/LibrePCB/issues/290

closed 2018-07-20 by ubruhin