LibrePCB/librepcb-rfcs #22: Author name detection: Use git name

dbrgn, 2018-07-17:
Right now on Linux we check for the author name using [getpwuid](https://github.com/LibrePCB/LibrePCB/blob/8773faf0d2a08020e3f1b669db9847e8e19f5e86/libs/librepcb/common/systeminfo.cpp#L101-L107). If that fails, we fall back to the current username.

Why don't we use the git commiter name (`git config --global user.name`) as second level of full name detection?

ubruhin, 2018-07-18:
Yeah as long as it is implemented in a safe way (not introducing unexpected behaviour for example if git config cannot be read) and preferably doesn't require adding more libraries as dependencies.

Another option would be to allow defining the user name in the workspace settings. This would make it possible to use a specific user name only for LibrePCB. Then the user name would be independent of the operating system or installed tools (like Git). The git config could still be used as fallback if no name is specified in the workspace settings.

dbrgn, 2018-07-19:
> Another option would be to allow defining the user name in the workspace settings. This would make it possible to use a specific user name only for LibrePCB.

I like that idea. Currently I manually have to update the author name of every library element when creating it.

ubruhin, 2018-07-19:
> I like that idea. Currently I manually have to update the author name of every library element when creating it.

You could set it once with `usermod` :wink:

hephaisto, 2018-07-19:
I really dislike when metainformation from the system is included in files without the users knowledge. It should be obvious and easy to disable or set manually.
I have a lot of projects where I don't want my name (or nickname which is easy to connect with my real name) connected with them.

ubruhin, 2018-07-19:
@hephaisto Sure, I don't like this too. But LibrePCB doesn't include such information *without* the users knowledge. It's always visible in the GUI if such system information is used, for example when creating a new project the "Author:" text edit field is just pre-filled with the user's name. The user is free to clear or modify the content of the Author field.

Storing this information in the workspace settings would probably be the best solution:

- The workspace creation wizard should ask the user what name will be used for all the "Author" fields (this field should be pre-filled with the automatically detected username to avoid unnecessary typing).
- This name is then stored in the workspace settings.
- The automatically detected username is never used again. Only the name stored in workspace settings is used from now on.

dbrgn, 2018-07-19:
> You could set it once with usermod :wink:

Haha, funnily enough this never occurred to me :smile: 

I agree with the current notion that the default user info should come from the workspace settings.

ubruhin, 2018-09-04:
Ready to be implemented: https://github.com/LibrePCB/LibrePCB/issues/322

ubruhin, 2018-10-03:
Implemented in https://github.com/LibrePCB/LibrePCB/pull/324.

closed 2018-10-03 by ubruhin