LibrePCB/librepcb-rfcs #23: Alternate pin functions

dbrgn, 2018-07-18:
Are there any conventions on what alternate pin functions to list in a symbol?

For example, for the blue pill:

[image]

...I used the following descriptions:

[image]

However, the selection is quite arbitrary. For example, I included SPI and I²C but not CAN.

I wonder if we could solve this using either a convention, or even in the library format somehow. What if every pin could contain alternate functions that are tagged with the subsystem somehow? Then maybe someone developing a project that uses I²C and UART will see the corresponding pin functions, but someone working on a project that uses the CAN bus will see only those alternate functions instead.

ubruhin, 2018-07-18:
Yeah maybe it would make sense to introduce a new concept for alternate functions. But keep in mind that this would be on *Component* scope, not on *Symbol* scope.

For the moment I would go with either completely without alternate function (just name pins "PA0", "PA1" and so on) or maybe with slashes as separator (e.g. "PA0/CTS2/WKUP")...

hephaisto, 2018-07-18:
There is already a similar system available: Having multiple symbol variants. Maybe it would be suitable to modify this system a bit? I was thinking about having e.g. a (SCK+MISO+MOSI)-Symbol that references three pins, but the same pins could also be referenced by a (GPIOA1+A2+A3)-Symbol.
The current system isn't flexible enough for that, though.

ouabache, 2018-10-25:
You should also consider that with FPGA's the chip designer doesn't pick the pinout. They give the PCB designer a package and a list of pads and the board designer picks the pins. LibrePCB could develop a tool that let the board designer create the UCF file for the FPGA design.

John Eaton

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/alternate-pin-functions/34

closed 2018-12-14 by ubruhin