LibrePCB/librepcb-rfcs #25: text placeholder that gets filled with the component name

SaculRennorb, 2018-08-20:
nothing more to say ^^

ubruhin, 2018-09-04:
Could you explain it more detailed please? I don't understand what you mean exactly ;)

SaculRennorb, 2018-09-04:
basically there is a placeholder for the component 'name' (`{{NAME}}`) but that just gets replaced with the abbreviation for the component plus the occurrence number eg. 'C1'
wouldn't it be good to have a placeholder that gets replaced with the name of the component int the library? a placeholder that gets replaced with for example 'PNP transistor' (the component name). I wouldn't want to use the default value for that since that gets problematic if you want to also have it for something.

ubruhin, 2018-09-04:
Ah, that's actually already possible :smiley:  You can use following placeholders:

```
{{COMPONENT}}
{{DEVICE}}
{{PACKAGE}}
```

In the schematic, this could then look like that:
[image]

Note: DEVICE and PACKAGE is currently only substituted if the component is added to exactly one board.

SaculRennorb, 2018-09-23:
oh, ok i thought i tried that and it didnt work or something, but maybe i just made a mistake.

so let me suggest something else:
some info card somewhere that shows you/lists the placeholders you can use, since i wasnt even sure if i just misstyped something or that placeholder didnt even exist

... now i just hope that doesnt already exist somewhere aswell and i just mised it alltogether...

ubruhin, 2018-10-03:
> some info card somewhere that shows you/lists the placeholders you can use, since i wasnt even sure if i just misstyped something or that placeholder didnt even exist

Yep, that makes sense. I created a feature request for that: https://github.com/LibrePCB/LibrePCB/issues/342

I'll close this issue since the initial question is resolved ;)

closed 2018-10-03 by ubruhin