LibrePCB/librepcb-rfcs #26: Flatpak packaging

Alexander-Wilms, 2018-08-25:
I'm not sure if this or [LibrePCB/librepcb](https://github.com/LibrePCB/librepcb) is the proper place to file this request. I think it would be nice if LibrePCB could be provided as a [Flatpak](https://flatpak.org/)  via [Flathub](https://flathub.org/). Below is a simple manifest, which I used to build LibrePCB. It seems to work fine so far; you can test it like this:
```
flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
flatpak install flathub org.kde.Sdk 5.11
flapak-builder --force-clean --ccache --repo=librepcb-repo librepcb-build org.librepcb.LibrePCB.yaml
flatpak remote-add --no-gpg-verify librepcb-repo librepcb-repo
flatpak install librepcb-repo org.librepcb.LibrePCB
flatpak run org.librepcb.LibrePCB
```

Contents of org.librepcb.LibrePCB.yaml:
```
app-id: org.librepcb.LibrePCB
runtime: org.kde.Sdk
runtime-version: 5.11
sdk: org.kde.Sdk
command: librepcb
rename-icon: librepcb
rename-desktop-file: librepcb.desktop
rename-appdata-file: librepcb.appdata.xml
finish-args:
- --socket=wayland
- --socket=x11
- --share=ipc
- --device=dri
- --share=network
- --filesystem=home # required for creating the workspace, downloading etc.
cleanup:
- /include
modules:

- name: librepcb
  buildsystem: qmake
  builddir: true
  sources:
  - type: git
    url: https://github.com/LibrePCB/LibrePCB
  - type: file
    path: librepcb.appdata.xml
  post-install:  
  - install -Dm 644 /app/share/pixmaps/librepcb.svg /app/share/icons/hicolor/scalable/apps/librepcb.svg
  - install -Dm 644 ../librepcb.appdata.xml /app/share/metainfo/librepcb.appdata.xml
```

You also need the appdata file from https://github.com/LibrePCB/LibrePCB/issues/314

ubruhin, 2018-08-26:
> I think it would be nice if LibrePCB could be provided as a Flatpak via Flathub.

Yes, that would be great :)

> Below is a simple manifest, which I used to build LibrePCB.

Nice, thanks! :+1:

Are you interested in resolving https://github.com/LibrePCB/LibrePCB/issues/313 and https://github.com/LibrePCB/LibrePCB/issues/314? Then I could follow the [App Submission](https://github.com/flathub/flathub/wiki/App-Submission) instructions.

ubruhin, 2018-08-26:
Hmm okay, I think we could provide our nightly flatpaks directly on https://download.librepcb.org. And as soon as we have stable releases, we can submit them to the official Flathub.

ubruhin, 2018-08-31:
I was able to use GitLab-CI to automatically build the flatpak: https://gitlab.com/LibrePCB/librepcb-flatpak/.

The bundle is hosted at https://download.librepcb.org/nightly_builds/master/librepcb-nightly-linux-x86_64.flatpak and the repository at https://download.librepcb.org/flatpak_repository/.

Usage:

```
flatpak remote-add --no-gpg-verify librepcb https://download.librepcb.org/flatpak_repository/
flatpak install librepcb org.librepcb.LibrePCB
flatpak run org.librepcb.LibrePCB
```

@Alexander-Wilms Could you try if it also works for you?

ubruhin, 2018-08-31:
Unfortunately the mimetypes are not installed properly...

Alexander-Wilms, 2018-08-31:
In can try installing it in a few minutes. I think anything that should be exported needs to have the org.librepcb.LibrePCB Präfix.

Alexander-Wilms, 2018-08-31:
Seems like you need to sign the build:
```
flatpak install librepcb org.librepcb.LibrePCB

Installing in system:
org.librepcb.LibrePCB/x86_64/master librepcb 10dcfbd090a0
  permissions: ipc, network, wayland, x11, dri
  file access: home
Is this ok [y/n]: Is this ok [y/n]: y
Installieren: org.librepcb.LibrePCB/x86_64/master von librepcb
Fehler: Failed to installieren org.librepcb.LibrePCB/x86_64/master: Can't pull from untrusted non-gpg verified remote
```

Alexander-Wilms, 2018-08-31:
Thanks for the repo, it works fine here. Only thing to note is that one needs to run `flatpak install` with sudo, since the build isn't signed: https://github.com/flatpak/flatpak/issues/1015

ubruhin, 2018-11-25:
I submitted LibrePCB 0.1.0 to Flathub: https://github.com/flathub/flathub/pull/738

ubruhin, 2018-11-28:
Package is now available at Flathub :smiley: https://flathub.org/apps/details/org.librepcb.LibrePCB

closed 2018-11-28 by ubruhin