LibrePCB/librepcb-rfcs #27: Governance model for LibrePCB development

rnestler, 2018-09-03:
I would like to start a discussion about the [governance model](https://opensource.guide/leadership-and-governance/) used for the development of LibrePCB. I'll start with trying to describe the current model and propose some formalizations and changes for future development.

## Current Situation (BDFL - model)

Currently @ubruhin does a great job at not only being the main developer of LibrePCB since more than 5 years, but also at administrating all the GitHub repositories, reviewing most of the MR, handling the issues and managing the LibrePCB website!
The LibrePCB Twitter account is managed by me and @dbrgn (which got me the unofficial title as "community manager" of LibrePCB by @ubruhin :wink:)
So most of the governance and administration is done by @ubruhin. This is partially because he is heavily invested in the project, but also partially because others don't have access to actually *do* any of these tasks.
For example I noticed, that the librepcb.org website doesn't properly support https (https://github.com/LibrePCB/LibrePCB.github.io/issues/17) but couldn't fix it, because I didn't have access.

## Proposed Changes / Ideas

Keep @ubruhin as a BDFL (benevolent dictator for live) since he does an awesome job at running the project!
Define a core team which has access to the GitHub organization and repos (no merging, since nobody besides @ubruhin has enough knowledge to decide that in my opinion), the server, the Twitter account and can do every administrative task. This can free up time for @ubruhin do actually develop the LibrePCB software.
This team should be able to share credentials for accessing the different accounts and servers.
Of course members of the core team shouldn't do stuff completely by their own without informing the rest of the core team (via the LibrePCB chat or other communication channels).

ubruhin, 2018-09-04:
Generally I like the idea, but I also wonder if that would really help. Most of the open issues can be resolved without administrative access, there are only very few things which need administrative access. For me it's OK to do *all* the administrative things, if I get help on the other issues ;)

There is also some additional risk if many users have administrative access (which is not even needed most of the time). Just as an example, the Gentoo GitHub organization [was hacked just a few months ago](https://wiki.gentoo.org/wiki/Project:Infrastructure/Incident_Reports/2018-06-28_Github). Basically it would also be possible to give administrative access only temporary when it's needed.

dbrgn, 2018-09-04:
I haven't been involved in the creation of this issue, but single-point-of-failure is also a risk. It would be good if at least a second person would have administrative access.

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/governance-model-for-librepcb-development/27

closed 2018-12-14 by ubruhin