LibrePCB/librepcb-rfcs #28: Using custom icons / themes

vini2003, 2018-10-03:
While decent-looking and easy to use, LibrePCB could make good use out of customizable icons, or at least different themes.
I've looked through a few icon packages as of late and found some interesting ones with fairly permissive licenses, such as the ones provided by [Icons8](https://icons8.com/license/), which are free to be used by open-source projects and very well made.
I made a comparison between default and custom icon versions of one LibrePCB window, which can be seen below:
[Default]
[image]
[Custom]
[image]
While not perfect, it looks much cleaner to me, and I'd appreciate being able to have a theme such as that one.
In short, it would be nice to be able to change icons.

0xB767B, 2018-10-03:
Maybe user-customizable is out of scope. But using a more pretty iconset by default could be an option to achieve a modern gui look.

hephaisto, 2018-10-03:
I would always prefer to have the icons I am used to, i.e. those supplied by one of the major frameworks. The [Default] screenshot looks very familiar, the one below does not (I just tried, and I guessed some of the functions wrong). Of course, the same will probably be true for @vini2003 the other way round.
My conclusion: Having customisable themes is nice to have, but changing the default icons to "achieve a modern look" does not sound good for me.
@vini2003 how much work was necessary to change the icons? Did you have to change any source files? If not, I would say, the feature is already present.

vini2003, 2018-10-03:
My bad, I should have mentioned I didn't actually change the icons in the program itself. I firstly tried to do that, but after trying to compile the program with the icons multiple times and failing, I gave up and edited the screenshot in order to fit the icons there.
I agree with @0x

ubruhin, 2018-10-03:
Actually Qt already supports icon themes (see [`QIcon::fromTheme()`](http://doc.qt.io/qt-5/qicon.html#fromTheme)), although only on Linux it works out-of-the-box. But maybe it's not too hard to also support it on Windows and macOS.

I also think it wouldn't be worth to support user defined themes, but supporting themes of desktop environments would be nice to make LibrePCB looking as native as possible.

vini2003, 2018-10-03:
Supporting default environment themes would be a good idea. I'm using it on Windows, so I don't know what the default theme here would look like, if there's one at all, but for Linux that would be pretty good.
I'd still like user-made themes, but if that's too much work, environment-based ones would suffice.

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/using-custom-icons-themes/26

closed 2018-12-14 by ubruhin