LibrePCB/librepcb-rfcs #30: Combining schematic and library editors into one

ouabache, 2018-10-18:
Is there a reason why LibrePCB has two separate programs for these editors? They have a huge overlap between them and some of the features in the library editor would also be useful to have in the schematic editor. Most schematic editors have the ability to draw graphics like lines,circles and shapes but LibrePCB only has lines. Both editors should have these.

Combining them into one would help it when you add design hierarchy and you have to sync schematic sheets with their symbols.

John Eaton

ubruhin, 2018-10-18:
> Is there a reason why LibrePCB has two separate programs for these editors?

Actually these aren't two programs, these are just two windows in the same program ;) One reason for having separate windows is to clearly distinguish between library management and project management.

> They have a huge overlap between them and some of the features in the library editor would also be useful to have in the schematic editor.

Maybe it looks like there is a huge overlap, but technically there is a big difference between a symbol and a schematic. Using separate windows helps to reduce the complexity of the code and was much easier to implement. At least with the current software architecture - maybe this changes some time (for example when adding support for hierarchical sheets, as you already mentioned).

> Most schematic editors have the ability to draw graphics like lines,circles and shapes but LibrePCB only has lines. Both editors should have these.

Actually I just didn't implement circles and polygons because they have very low priority ;) I think it shouldn't be too hard to add support for these shapes in schematics. But it's not just a matter of the editor, also the underlying data structure currently misses support for such shapes.

ouabache, 2018-10-19:
One reason for having separate windows is to clearly distinguish between library management and project management.
---------------------------------------------------------------------------------------------------------------------
When you implement design hierarchy there won't be any differences. You projects could be someone's library and your libraries were someone's projects. Some may be both in your own designs.

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/combining-schematic-and-library-editors-into-one/24

closed 2018-12-14 by ubruhin