LibrePCB/librepcb-rfcs #31: Symbol Pins

ouabache, 2018-10-23:
The symbol pins should have an "Invisible" property for when the pin name is not needed and would
only clutter up the schematic. This is common with digital gates.

You should also consider moving the symbol pin into a library that can be attached to either a symbol body or schematic. gEDA did this with bus rippers and  it will make it easier to deal with pins,ioports, bus spilitters and global symbols with having to rewrite code.

John Eaton

ouabache, 2018-10-24:
Found the component section and it works. Thank you.

Yes it is the same as #19, I am still lobbying for it.

ubruhin, 2018-10-24:
OK, so I'm closing this (#19 stays still open).

closed 2018-10-24 by ubruhin