LibrePCB/librepcb-rfcs #32: 3D models of packages and boards

ubruhin, 2018-10-25:
LibrePCB packages should have support for 3D models, so that we can implement a 3D board viewer & export.

In KiCad, footprints can have two different types of 3D models:
- [VRML](https://en.wikipedia.org/wiki/VRML) for 3D board viewer and raytracing
- [STEP](https://en.wikipedia.org/wiki/ISO_10303) for export to mechanical CAD

Although it's a mess to have two different file formats for 3D models, I guess this is the most reasonable solution. So I think it would make sense if we do it the same way...

Or are there other opinions?

dbrgn, 2018-10-25:
VRML is horrible unfortunately... But I don't really know a good alternative.

On the other hand, using the same format as KiCAD would allow us to use their 3D models and vice versa. With the growing popularity of KiCAD, maybe better editing tools will pop up. KiCAD libraries are licensed CC-BY-SA, so we could use them. This is probably a case where converging on a common format would give the open source community a lot of benefits.

Also, it seems that nowadays FreeCAD can generate valid VRML files for KiCAD. That wasn't the case a few years ago :)

ubruhin, 2018-10-26:
> On the other hand, using the same format as KiCAD would allow us to use their 3D models and vice versa.

Yes, that *would* be really helpful!

> KiCAD libraries are licensed CC-BY-SA, so we could use them.

Hmm I'm not sure if we could/should use them. Actually I want to have all official LibrePCB libraries to be published under CC0, so that users can do absolutely *everything* with the libraries, without even worrying about their license. CC-BY-SA breaks this, as it has restrictions :(

dbrgn, 2018-10-26:
It's a bit special, see https://github.com/KiCad/kicad-packages3D/blob/master/LICENSE.md. Their license would allow *users* of LibrePCB to use the files without having to attribute the copyright owner.

ubruhin, 2018-10-26:
But as far as I understand, users of LibrePCB have to keep the CC-BY-SA license if they *redistribute* our libraries (containing the 3D models).

It's a mess. I *really* don't want to have such restrictions on our libraries, but without KiCad's 3D models it's very hard to get good 3D models :sob:

ubruhin, 2018-10-28:
Hmm actually I think for the beginning it would be sufficient to support only STEP models. Compatibility with mechanical CADs is more important than realistic 3D previews, and 3D previews with STEP models look already pretty nice.

If really needed, we can still add support for VRML (or something else) some time later (e.g. for raytracing to get realistic looking 3D previews).

dbrgn, 2018-10-28:
As long as we have the option to add more formats later on, I guess that's fine.

ubruhin, 2018-10-28:
> As long as we have the option to add more formats later on, I guess that's fine.

Sure, I think that should be possible.

I just wonder if there is really no 3D file format which allows both, export to MCAD and realistic previews with raytracing (incl. textures etc.)... Would be great to have only one format for everything. I think about something like the Povray or OpenSCAD file format. They are especially cool because they can be parameterized (good for library generators!). But at the end we need a STEP export, and I'm not sure if that's possible with these formats.

There must be some innovative ideas for this - just using STEP and VRML models is boring :grin:

dbrgn, 2018-10-29:
OpenSCAD cannot export STEP files, and styling support for surfaces/materials is nonexistent. Although it would be awesome if we could somehow directly support OpenSCAD source files :slightly_smiling_face:

rnestler, 2018-10-29:
> just using STEP and VRML models is boring grin

[Boring](http://mcfunley.com/choose-boring-technology) may be a good thing.

ubruhin, 2018-10-31:
> OpenSCAD cannot export STEP files, and styling support for surfaces/materials is nonexistent. Although it would be awesome if we could somehow directly support OpenSCAD source files slightly_smiling_face

Yes I know, I just mentioned OpenSCAD as an example for what I mean :)

Actually the PovRay format should support everything we would need. But I think it's also quite complex, and probably it would require high effort to integrate it in LibrePCB (for both, 3D preview and STEP export).

> Boring may be a good thing.

I agree, but having STEP *and* VRML is not only boring for us developers, but also cumbersome for users because packages need 3D models in two different file formats containing *exactly* the same 3D model (just because STEP is missing textures, and VRML can't be used for MCAD). That's an incredible amount of duplicate data which could be avoided if there is a file format which allows both, realistic 3D preview *and* export to STEP.

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/3d-models-of-packages-and-boards/23

closed 2018-12-14 by ubruhin