LibrePCB/librepcb-rfcs #33: Add text functionality to schematic editor

dbrgn, 2018-11-01:
Right now text cannot be placed in the schematics. Would be nice to have that.

## Desired functionality

- Place text blocks on the schematics
- Allow customizing font size and weight

## Brainstorming

- Allow markdown formatting and only render on save (simplifies implementation)
- Should font selection be per-block or for the entire sheet?
- Format classes (heading1, heading2, text, small) would be nice to be able to achieve consistent formatting

ouabache, 2018-11-02:
Yes for Text. 
Select font,size and color.
Do not place on snap grid,You need a finer adjustment for location.

If you do place on snap grid then allow for the 9 placement modifiers: above left,above,above right,left ,cent etc.

John Eaton

ubruhin, 2018-11-03:
Support for text in schematics is needed for sure. But I'm not sure about Markdown because it's very limited and many EDA users probably don't know Markdown and thus aren't comfortable with it. But I would agree to use Markdown just as an optional, alternative input method for texts.

Qt would support a [subset of HTML](http://doc.qt.io/qt-5/richtext-html-subset.html) out of the box. With that it shouldn't be too hard to support rich text in LibrePCB...

Choosing fonts is tricky. Actually only fonts bundled with LibrePCB should be used, the use of system fonts would break portability of LibrePCB projects.

dbrgn, 2018-11-04:
> I guess this is pretty easy with Qt.

Ah, I wasn't aware of that :slightly_smiling_face: I guess you would then store the text as HTML inside the schema file?

Does Qt allow passing in a whitelist of HTML tags and attributes allowed?

> So maybe we should allow the user to choose what kind of text they want to place on schematics

Yeah, I think that would be good: Three different types of text elements.

I could imagine a "add text" button in the toolbar that has a context menu where you can choose which variant to add (and then the chosen variant becomes the new default). Not sure if that's doable with Qt out of the box.

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/add-text-functionality-to-schematic-editor/19

closed 2018-12-14 by ubruhin