LibrePCB/librepcb-rfcs #34: Gate and pin swapping

ouabache, 2018-11-04:
I am trying to understand how LibrePCB will do gate and pin swapping. Let me know if I have this right.

I create a symbol for a NAND2 gate.
I create a symbol for a power block.
I create components for a NAND2 and power block 

I put a bunch of NAND2's on a schematic with one power block for every 4 gates.

I create a component for a 74LS00 with 4 NAND2's and a power block.

I select 4 NAND2 components and one power block and bundle them into a 74LS00.

I can now select any two NAND2 gates from any 74LS00's and LibrePCB can swap them.

So what about pin swapping?

I create another NAND2 with swapped pins and make it a symbol variant for my NAND2 component.

Swapping pins is simply a case of selecting a gate and choosing its other symbol variant.
 
But if I want to swap pins after packaging I need to create 16 variants for the 74LS00 representing every possible combination of normal and swapped pins.

Is this how it will work?

John Eaton

ubruhin, 2018-11-04:
I don't know exactly how it will be implemented. Maybe something like that:

- You create a symbol for a NAND2 gate
- You create a symbol for a power block
- You create components for every count of gates:
  - "NAND2 Single"
  - "NAND2 Dual"
  - "NAND2 Quad"
- You create devices for every "real" part, e.g. "74LS00" for the component "NAND2 Quad".

The components need only one symbol variant containing all gates. These gates may be assigned to a "swap group" to make them swapable.

In schematics you place the "NAND2 Quad" component with all 4 gates + power block. As the gates are marked as swapable, these can be swapped in schematics.

Pin swapping may be implemented with swap groups at component signal level. Every signal can be swapped with other signals contained in the same swap group.

But that's just a rough idea... ;)

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/gate-and-pin-swapping/22

closed 2018-12-14 by ubruhin