LibrePCB/librepcb-rfcs #35: Automatic synchronisation between schematic and board

marmei, 2018-11-04:
##### SUMMARY
Are you panning to keep the automatic synchronisation between the schematic and the board? I know that major PCB design tools are not using such a way because of several issues. Especially in Eagle which I used sometimes previously, the board and schematic lost consistency and forward and backannotation got lost.
In fact if you are planning to implement hierarchical sheets of if one is willing to performs copy and paste of components or entire schematics, such automatic synchronisation may cause frustration to the user.

It also does not make the schematic transparent from the PCB.
Will advanced features, like SPICE netlists than update automatically?    

https://www.altium.com/documentation/16.1/display/ADES/((Design+Synchronization))_AD

##### SOLUTION
Create a standalone console application / script which syncs the board and the schematic or a 3D view.
This could be done also automatically within your application but may also allow to load KICAD/ALTIUM/EAGLE/etc. netwlist into your pcb - or create netlists to sync with the schematics.
Do you think it is possible to implement such tool independently for the main LibrePCB source code (as a submodule)

ubruhin, 2018-11-04:
> Are you panning to keep the automatic synchronisation between the schematic and the board?

Generally, yes. IMO automatic synchronization makes designing PCBs much easier. Especially for hobbyists I think the manual synchronization can be very cumbersome. I even had several issues with the forward annotation in Altium. So I think both concepts have their advantages and disadvantages, and at this time I'd like to go with automatic synchronization.

> I know that major PCB design tools are not using such a way because of several issues. Especially in Eagle which I used sometimes previously, the board and schematic lost consistency and forward and backannotation got lost.

That's an issue of how it's implemented in Eagle. In LibrePCB this issue doesn't exist, it's not possible to loose the consistency.

> In fact if you are planning to implement hierarchical sheets of if one is willing to performs copy and paste of components or entire schematics, such automatic synchronisation may cause frustration to the user.

I agree that this will be a tricky thing. But I think some issues can be avoided with a smart implementation. For example cut/paste in schematics could be implemented in a way which doesn't change anything on the board, as long as the netlist after pasting is identical as before.

> It also does not make the schematic transparent from the PCB.

Why does it have to be transparent? In the end, both should represent exactly the same netlist. With automatic synchronization, the netlist is *guaranteed* to be always the same. With manual forward/backward annotation, schematic and board can be out of sync, representing different netlists.

> Will advanced features, like SPICE netlists than update automatically?

Of course that's the plan :)

> Create a standalone console application / script which syncs the board and the schematic or a 3D view. This could be done also automatically within your application but may also allow to load KICAD/ALTIUM/EAGLE/etc. netwlist into your pcb - or create netlists to sync with the schematics.
Do you think it is possible to implement such tool independently for the main LibrePCB source code (as a submodule)

This would completely break the fundamental concept of automatic synchronization. You can also create importers for KiCad/Altium/Eagle without removing automatic synchronization. Theoretically you could also import a netlist and a board, but without schematics, if you want a board-only project. LibrePCB stores the netlist independent of schematics and boards: https://github.com/LibrePCB/LibrePCB/blob/master/dev/diagrams/svg/circuit_overview.svg

marmei, 2018-11-04:
Is it than possible to have a schematic in LibrePCB without a board?
Do you think LibrePCB will be capable to work in team? One creates the schematic, the other one a board and vice versa.
Or think of the use case -

Person A: changes some board wiring and pushes his pcb to github.
Person B: Updates some components in the schematics pushes to github.
It might be possible that A broke B's schematic and B roke A's board.
What do you think?

marmei, 2018-11-04:
Thank you very much for your answers, which makes it clear for me.
I have one last question.

I realize now that the schematic and the board file are somehow bound together. And so it will break consistency if they are edited independently. 

Does it make sense to combine both files into one single file, such that schematic and board are "married" together? This has than the advantage that the user can not break things (e.g. by forgetting to commit the schematic but board changes are committed).

ubruhin, 2018-11-04:
> I realize now that the schematic and the board file are somehow bound together.

Not directly - There is always the so-called "circuit" (which contains the netlist) in between. This is also saved in a separate file. Any change in schematics can be one of following cases:
1. Only the schematic file is affected (graphical-only changes, like moving or rotating symbols)
2. Only the circuit file is affected (netlist-only changes, like changing the value of a resistor)
3. Schematic and circuit files are affected (combinations of 1. and 2.)

In variant 2. and 3. it's also possible that the board file is affected. For example if a component was removed from the schematic, it also gets removed from the circuit and from the board.

> Does it make sense to combine both files into one single file, such that schematic and board are "married" together?

I don't think so. Having multiple files helps to keep track of what changes are made in which parts of a project. One single file would be extremely large and thus makes reviewing diffs much harder.

> This has than the advantage that the user can not break things (e.g. by forgetting to commit the schematic but board changes are committed).

This could be solved with a continuous integration system :) If you forgot to commit a board file, which would then break the synchronization between schematics and board, LibrePCB rejects to open that project because it's broken. With the [LibrePCB CLI](https://docs.librepcb.org/#cli) you can check the validity of a LibrePCB project on a continuous integration system, so the developer gets notified about his mistake just a few seconds or minutes after pushing his commit ;)

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/automatic-synchronisation-between-schematic-and-board/18

closed 2018-12-14 by ubruhin