LibrePCB/librepcb-rfcs #36: Selecting Thru hole vs surface mount

ouabache, 2018-11-05:
It appears that you must select between TH and SM when you place a component on the schematic.  You don't always know that when you are creating the schematic. I will do a lab prototype board using TH parts and individual resistors but switch to SM and resistor packs for production. I want one schematic that covers both boards.  How will that work?

John Eaton

ubruhin, 2018-11-05:
> It appears that you must select between TH and SM when you place a component on the schematic.

Hmm actually not :) What component are you talking about?

Maybe you are confused because you *can* choose the exact device, but that's optional?

ouabache, 2018-11-06:
I might be confused. I saw the options to select the devices but didn't realize that you could still select the component and not the devices. That will solve my problem.

Thanks

ouabache, 2018-11-06:
I think whats throwing me for a loop is that you have one set of categories for symbols,components and devices. Packages have their own set.

I want to create a digital library so I create a cmpcat for Logic followed by Combinatorial and Registered followed by And Or Nor Nand etc followed by default and demorgan.

I fill this with all my symbols and components but when I go to select an AND2 gate I find it in Logic/Combinatorial/And but I also see selections for default and demorgan. These are not real because I make that selection as symbol variants in the AND2 component. I would rather not see them
at all because they only confuse things. 

My original problem was that I wanted to select a component but when I reached it I saw two additional
selections for devices that used that component. If I hadn't seen them I would have pick the the correct component but missed it because two devices appeared on the menu.

I have started creating top level cmpcats for symbol,component and devices to avoid this issue. You wind up with three separate trees but it keeps out the selections that you don't want to see.

John Eaton

ubruhin, 2018-11-06:
> I might be confused. I saw the options to select the devices but didn't realize that you could still select the component and not the devices. That will solve my problem.

:+1: :smiley: 

> I fill this with all my symbols and components but when I go to select an AND2 gate I find it in Logic/Combinatorial/And but I also see selections for default and demorgan. These are not real because I make that selection as symbol variants in the AND2 component. I would rather not see them
at all because they only confuse things.

I'm not sure if I understand correctly. Is the problem only that you can see categories which only contain symbols, but no components? If yes, I planned anyway to hide categories which do not contain components (when looking for components) or symbols (when looking for symbols). Would that solve your issue?

ouabache, 2018-11-07:
Yes,

Hiding the categories that do not contain a relevant selection would work. You would never see a selection unless it contained an appropriate selection.

The same thing might  also work with  package selection. You could merge the packages in with the rest and only display the selections that contained packages.