LibrePCB/librepcb-rfcs #37: Marketing Tip

ouabache, 2018-11-07:
One problem that Kicad has is that there are a lot of videos and app notes out there showing how to use it. This is great until they do a new release and most of them are obsolete but still show up in search results.

Ubuntu does lots of releases and tags every new one with a nickname. If you google "ubuntu bionic beaver" your results will only pertain to that release.

I would suggest the LibrePCB adopt a similar strategy and include a nickname along with every release to act as a keyword for searches.  Promote this when you have a first release and it will save you a lot of trouble when Rev 2 comes along.

GIven LibrePCB's origin I would suggest that you use the names of Swiss mountains for your nicknames.

rnestler, 2018-11-08:
> GIven LibrePCB's origin I would suggest that you use the names of Swiss mountains for your nicknames.

Like almost every Swiss company I know does for their internal project names :wink:

Nevertheless, I like the idea.

ubruhin, 2018-11-10:
Hmm the idea is interesting, but to be honest, I'm not sure what I should think about it ;)

> Ubuntu does lots of releases and tags every new one with a nickname. If you google "ubuntu bionic beaver" your results will only pertain to that release.

True, but I always have the opposite problem - If I read something about e.g. Ubuntu Bionic Beaver, I'm never sure which version it is. Is it 17.10? 18.04? 18.10? But if the version number is written, I know which one is meant :wink:

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/marketing-tip/39