LibrePCB/librepcb-rfcs #38: Pin Attributes

ouabache, 2018-11-11:
If you compare LibrePCB's pin attributes with Kicads you see that we have

Pin Name
Position x,y
Length
Rotation

Kicad has

Pin Name
Position x,y
Length
Rotation
Pin Number
Pin Name text size
Pin Number text size
Unit
Convert
Etype  (in,out bidi etc)
Shape  clocked,inverted etc

Pin number is handled in the device so we do not need it and the Unit,Convert are handled in the component so we don't need them either.

Invisibility is also handled in the component while kicad does it in the Shape.

Kicad can only do three things that we can't and those are:

Resizing Pin Name text
Attaching an Etype attribute
Using an alternate graphic

LibrePCB should consider adding these if we want to be able to share designs with kicad.

There are also some attributes that neither one has that would be really useful to add.

1) Pin Name Position. You always want to have the option to move the Pin Name in respect to its origin. This should not be on the snap grid.

2) Display Name. The Pin Name is used by about half a dozen different engineering groups and they all have their own pin naming requirements. The Pin Name must meet all of them. It is impossible to meet that requirement. When I worked for HP our board designers got tired of hearing that they couldn't use a certain pin name that they wanted because it was also also a VHDL keyword that they gave up and implemented a display name. 

The pin names that you saw on our schematics were simply graphics that the board designers wanted and were not connected in any way with the actual pin name.

We did a lot of chips with 3 or 4 different functions on each pin that were used in 3 or 4 different products. Display Names meant that each design could display the name that was important to them and did not have to compromise because some other group using the chip had other needs.

3) Indexes  Low and High.  When you implement signal vectors and hierarchy then you will need these.

4) Hierarchy.  All pins connect to a different level in the hierarchy. Some go up and others go down. You need an attribute for this. Sometimes the same pin will do both in different components. A pin on a FPGA will connect down the hierarchy while the same pin on an off board connector will connect up. An attribute lets you use one pin for both cases.

5) ????????  Don't kid yourselves into thinking that you can predict all the attributes that the users will need in the future. Provide a method so that the user can add their own attributes and set them from an attribute in the component. You do not want the make the user wait until the next release for a feature that they need today

John Eaton

ubruhin, 2018-11-12:
> Resizing Pin Name text

Is this really needed? Do you have a concrete example where it is needed?

> Attaching an Etype attribute

This is the electrical function of the pin (used for ERC), right? In the file format this is already supported, but it's implemented for component signals, not for pins (see "role" property):

```scheme
 (signal 343a05e0-6e9b-4317-a7de-61bd3c774b5c (name "C") (role passive)
  (required true) (negated false) (clock false) (forced_net "")
 )
```

It's just hidden in the UI because it is not used by the ERC yet ;)

> Using an alternate graphic

You mean for clock/invert?

> Pin Name Position.

Yes, I agree that this would be useful. Also rotation, not only position.

> Display Name.

I guess you mean https://github.com/LibrePCB/librepcb-rfcs/issues/23?

For the things related to hierarchical sheets, we should wait until there is a concrete concept how hierarchic sheets will be implemented exactly...

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/pin-attributes/41