LibrePCB/librepcb-rfcs #40: Reverse engineering mode

Harvie, 2018-11-29:
If i try to create connection in "Board editor" window and that connection is not already in schematic, it gives me following error:

[image]

And that is perfectly OK. But what if i am reverse engineering the board. I have photos of board and i want to recreate the same board in Board editor and turn that into a schematic. (given that i've already added all the components to the schematic). That would help me to untangle the connections on the board to usable schematic...

I still think this should not be enabled by default. Surely i don't want to screw with the schematic by accidentaly adding new connections when making board from schematic. But having option to enable this mode (with proper warning dialog) would be great.

ubruhin, 2018-11-29:
I have this use-case in mind since the beginning of LibrePCB because I also like to have this feature :) The core classes should be (more or less) ready for that feature, but the schematic and board editors aren't, so they have to be extended.

> I still think this should not be enabled by default. Surely i don't want to screw with the schematic by accidentaly adding new connections when making board from schematic. But having option to enable this mode (with proper warning dialog) would be great.

:+1:

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/reverse-engineering-mode/21

closed 2018-12-14 by ubruhin