LibrePCB/librepcb-rfcs #42: Community Platform

dbrgn, 2018-12-06:
I think it would be good to have some community platform besides our chat for asking questions and sharing progress. Something that is a bit more structured.

Two options come to mind: Reddit and Discourse.

## Reddit

+ Already has a big community, so it's easy for people to join a subreddit
+ No hosting required
+ Easy discoverability
- We don't control the data
- Less structured

## Discourse

- Must be self-hosted (but that's easy)
- People have to sign up (although they can use their GitHub login for example)
+ Structured, easily searchable
+ We control the data

dbrgn, 2018-12-06:
My favorite would probably be Discourse, but I guess it would be best if that wouldn't run on our API server. Which means that we don't currently have a server available, right?

Any ideas?

ouabache, 2018-12-06:
Do both. Use Reddit for users and Discourse for developers

ubruhin, 2018-12-06:
I would strongly prefer Discourse as it's much more structured.

Discourse [offers free hosting for open source projects](https://blog.discourse.org/2018/11/free-hosting-for-open-source-v2/), maybe that would be sufficient for us?

> Do both. Use Reddit for users and Discourse for developers

That implies that the only discussion paths are user<->user and developer<->developer. But in reality there are also many discussions between users and developers, so IMHO we should have only one platform for all kinds of discussions.

Btw, with a proper discussion platform I think this RFC repository is no longer needed...

dbrgn, 2018-12-06:
Woah, that offer looks great. I think we certainly qualify. And there's no lock-in, we can always move to self-hosted.

Could you apply?

dbrgn, 2018-12-07:
That's no problem, you can create separate subcategories, then developers can read just the developer discussions, not the user questions.

ouabache, 2018-12-08:
Sounds good. I'll support it.

John Eaton

ubruhin, 2018-12-10:
> Could you apply?

Done :-)

ubruhin, 2018-12-12:
Discourse is online! https://librepcb.discourse.group/

I think it would make sense to move all issues of this repository to Discourse and archive this repository. Or what do you think?

dbrgn, 2018-12-12:
I agree :slightly_smiling_face: I will move my own issues.

dbrgn, 2018-12-12:
You should probably start out by defining some categories.

(I signed up btw. My usual username.)

ubruhin, 2018-12-12:
There is a GitHub Issues to Discourse migration tool: https://www.discourse.org/github/

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/community-platform/16

closed 2018-12-14 by ubruhin