LibrePCB/librepcb-rfcs #43: Replacing package of a device

dbrgn, 2018-12-08:
Sooner or later, we will run into the situation where we used a generic footprint for a specific device, but then later notice that the footprint is not ideal and that we need a vendor-specific footprint.

Right now replacing the package is not supported in a reverse-compatible way, I think. Are there any ideas / plans on how to achieve that? Deprecating the entire device and replacing it with an almost identical new device doesn't sound ideal to me.

Maybe we could deprecate footprint variants and simply add the new package as a footprint variant?

ubruhin, 2018-12-09:
The problem is that this is not trivial to fix. The project library updater is currently extremely simple and it works pretty fine as long as there are no breaking changes in library elements (like changing the package of a device).

But as soon as there is a breaking change, the updater completely fails (no element can be updated at all). To fix that, the updater would need to be designed *much* more sophisticated because it would also need to modify boards to update all pad UUIDs of traces connected to them. To achieve this, the updater would need to analyze the difference between the old device and the new one to know which pads were replaced by which.

If we ever make the updater that sophisticated (if even possible), we can remove the requirement that devices must not switch to another package. But until then we must just replace outdated devices with new ones, otherwise we would frustrate users because the project library updater breaks all the time.

Btw, for devices it's actually fine to remove them instead of just deprecate them because there is no other library element which depends on devices (it's the "top level" library element). So for the moment we could avoid too many deprecated devices by just removing them ;)

But the general rule is always: Choose the proper package right after creating a new device. Think twice about if a generic package is fine or not.

dbrgn, 2018-12-09:
What about the idea with the footprint variants? Maybe that could be used as an upgrade path somehow.

ubruhin, 2018-12-09:
I don't know what you mean exactly. A device doesn't know anything about footprints, so it has no control over which footprints are used or not...

dbrgn, 2018-12-09:
Ah, I somehow had in mind that a device can have package variants, but that's not the case... Doesn't really make sense anyways :slightly_smiling_face:

ouabache, 2018-12-10:
What if you put something in the board file that said UUID xxxxx is now yyyyy for instance ZZ. Then any reference to the old UUID is redirected to use the new one instead? For example I put a resistor in a schematic but then when I do the board I "uplift" the UUID for an instance to point to a ResistorPack element. The ResistorPack uses the same symbol as the resistor so the only change you see is the instance and device names.

John Eaton

ubruhin, 2018-12-14:
This issue was moved to https://librepcb.discourse.group/t/replacing-package-of-a-device/20

closed 2018-12-14 by ubruhin